BrettRD/ros-gst-bridge #1: Unable to run

gc-robotics, 2020-12-04:
pipeline_node-3] (gst-plugin-scanner:19746): GStreamer-WARNING **: 13:12:04.710: Failed to load plugin '/root/r/install/gst_bridge/lib/gst_bridge/librosgstbridge.so': /root/r/install/gst_bridge/lib/gst_bridge/librosgstbridge.so: undefined symbol: gst_base_src_set_format
[pipeline_node-3] 
[pipeline_node-3] (python3:19727): GStreamer-WARNING **: 13:12:04.718: Failed to load plugin '/root/r/install/gst_bridge/lib/gst_bridge/librosgstbridge.so': /root/r/install/gst_bridge/lib/gst_bridge/librosgstbridge.so: undefined symbol: gst_base_src_set_format
[pipeline_node-3] [ERROR] [gst_pipeline_node]: plugin load error
[pipeline_node-3] [WARN] [gst_pipeline_node]: Missing gstreamer plugins: ['rosgstbridge']

Not sure what the problem is... the custom plugin doesnt appear in gst-inspect either...  any suggestion? thanks.

BrettRD, 2020-12-05:
Thanks for giving it a whirl!

It might be missing a dependency during build for librosgstbridge.so,
Does librosgstbridge.so exist? it should be in ```install/gst_bridge/lib/gst_bridge/```

It doesn't add the custom plugin to the system gstreamer path, so you need to tell gst-inspect and gst-launch where to find it using the ```--gst-plugin-path``` flag:
```gst-inspect-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ rosgstbridge```
```gst-inspect-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ rosaudiosink```

If the plugin is malformed, gst-inspect should be unhappy about them and should give us some more clues.

gc-robotics, 2020-12-05:
Thanks for the support, Brett.

I was not able to find rosgstbridge or rosaudiosink, but I could see that libgst_bridge and librosgstbridge are both blacklisted. Please, refer to the output below:
```
root@d03a46f6faa0:~/robotica/install# gst-inspect-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ rosgstbridge
No such element or plugin 'rosgstbridge'
root@d03a46f6faa0:~/robotica/install# gst-inspect-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ librosgstbridge.so
Plugin Details:
  Name                     librosgstbridge.so
  Description              Plugin for blacklisted file
  Filename                 /root/robotica/install/gst_bridge/lib/gst_bridge/librosgstbridge.so
  Version                  0.0.0
  License                  BLACKLIST
  Source module            BLACKLIST
  Binary package           BLACKLIST
  Origin URL               BLACKLIST

  0 features:

```

Also, this is the content of gst_bridge/lib:

ls -R
.:
gst_bridge  libgst_bridge.so

./gst_bridge:
librosgstbridge.so
`

Thanks

BrettRD, 2020-12-05:
Thanks for that, the blacklist probably means something went wrong during compile and the plugin can't be loaded
To show what other files gstreamer has blacklisted, you can run ```gst-inspect-1.0 -b```

libgst_bridge.so is a shared lib for other ROS nodes that provides conversions between data structures like OpenCV, we can ignore libgst_bridge.so

Lets try expanding the debug output from gst-inspect, this should tell us why it's blacklisting it:
In the folder that contains librosgstbridge.so:  ```GST_DEBUG=4 gst-inspect-1.0 ./librosgstbridge.so```

This should also show which versions of Linux, Gstreamer and Glib are in use (you might like to redact the machine name from the output)

gc-robotics, 2020-12-05:
Ok, so I followed your instructions and this is the output that was produced:

`GST_DEBUG=4 gst-inspect-1.0 ./librosgstbridge.so 
0:00:00.000163732 30251 0x5576601d3a00 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.14.5
0:00:00.000253458 30251 0x5576601d3a00 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/x86_64-linux-gnu
0:00:00.000306490 30251 0x5576601d3a00 INFO                GST_INIT gst.c:607:init_pre: Linux d03a46f6faa0 5.4.0-53-generic #59-Ubuntu SMP Wed Oct 21 09:38:44 UTC 2020 x86_64
0:00:00.000988315 30251 0x5576601d3a00 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.002223667 30251 0x5576601d3a00 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.002938777 30251 0x5576601d3a00 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.003187517 30251 0x5576601d3a00 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.003237384 30251 0x5576601d3a00 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.003329937 30251 0x5576601d3a00 INFO            GST_REGISTRY gstregistry.c:1727:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.040010258 30251 0x5576601d3a00 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.x86_64.bin in 0.036613 seconds
0:00:00.040126609 30251 0x5576601d3a00 INFO            GST_REGISTRY gstregistry.c:1583:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.042548160 30251 0x5576601d3a00 INFO            GST_REGISTRY gstregistry.c:1685:scan_and_update_registry: Registry cache has not changed
0:00:00.042579052 30251 0x5576601d3a00 INFO            GST_REGISTRY gstregistry.c:1762:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.042599735 30251 0x5576601d3a00 INFO                GST_INIT gst.c:807:init_post: GLib runtime version: 2.56.4
0:00:00.042614576 30251 0x5576601d3a00 INFO                GST_INIT gst.c:809:init_post: GLib headers version: 2.56.4
0:00:00.042626149 30251 0x5576601d3a00 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.054611152 30251 0x5576601d3a00 WARN      GST_PLUGIN_LOADING gstplugin.c:792:_priv_gst_plugin_load_file_for_registry: module_open failed: ./librosgstbridge.so: undefined symbol: gst_base_src_set_format

(gst-inspect-1.0:30251): GStreamer-WARNING **: 09:57:06.749: Failed to load plugin './librosgstbridge.so': ./librosgstbridge.so: undefined symbol: gst_base_src_set_format
Could not load plugin file: Opening module failed: ./librosgstbridge.so: undefined symbol: gst_base_src_set_format
`

it looks like the plugin itself doesnt find gst_base_src_set_format. Is that correct?

BrettRD, 2020-12-05:
> it looks like the plugin itself doesnt find gst_base_src_set_format. Is that correct?
That's right!  This is a linker stage bug, and I'm not sure why it's not affected my build.

It needs one more include at link stage
```
target_link_libraries(rosgstbridge PUBLIC
...
  ${GSTREAMER_BASE_LIBRARIES}
)
```

BrettRD, 2020-12-05:
That's in the ros2 branch now,
See how you go with latest: da80cc58a142edc5cdaa197f046cb33d9cb1a420

gc-robotics, 2020-12-05:
I tried and that problem has been fixed. Although, I am getting a similar build error on 

undefined symbol: gst_audio_info_to_caps

BrettRD, 2020-12-05:
> Is this the case of another path to be added for the linker?

Yes. (or possibly a clean build?)

On my machine, (ubuntu 20, gstreamer 1.16.2) that symbol is defined in
```/usr/lib/x86_64-linux-gnu/libgstaudio-1.0.so.0.1602.0```
Which is linked by  ```  ${GSTREAMER_AUDIO_LIBRARIES}```
There shouldn't be a significant difference going from gstreamer 1.14 to 1.16

Was that a clean build?  ```rm -rf build install```

What do you get looking for that symbol in the gstreamer lib folder? ```/usr/lib/x86_64-linux-gnu``` on my machine

```
$ grep -rins  "gst_audio_info_to_caps" /usr/lib/x86_64-linux-gnu
Binary file /usr/lib/x86_64-linux-gnu/libgstbadaudio-1.0.so.0.1602.0 matches
...
Binary file /usr/lib/x86_64-linux-gnu/libgstaudio-1.0.so.0.1602.0 matches
...
```

gc-robotics, 2020-12-05:
Yes, that was a clean build. 

These are the libs that I can find:

Binary file /usr/lib/x86_64-linux-gnu/girepository-1.0/GstAudio-1.0.typelib matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrtp.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstinter.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstinterleave.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstmxf.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrawparse.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstdv.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstpulseaudio.so matches
Binary file /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstjack.so matches
Binary file /usr/lib/x86_64-linux-gnu/libgstbadaudio-1.0.so.0.1405.0 matches
Binary file /usr/lib/x86_64-linux-gnu/libgstaudio-1.0.so.0.1405.0 matches
Binary file /usr/lib/x86_64-linux-gnu/libQt5WebKit.so.5.212.0 matches

I have found some other issues when building on Eloquent.
As soon as I have a working version, I'll submit a PR. 

Thanks again for the help!

gc-robotics, 2020-12-08:
I have rechecked out the ROS2 branch. I am getting some different issues. I will close this one. Thanks.

closed 2020-12-08 by gc-robotics