BrettRD/ros-gst-bridge #17: Clean build fails.

gc-robotics, 2021-04-03:
Starting >>> audio_msgs
Starting >>> gst_pipeline
Finished <<< gst_pipeline [1.28s]                                                
Finished <<< audio_msgs [8.84s]                     
Starting >>> gst_bridge
[Processing: gst_bridge]                             
--- stderr: gst_bridge                                
In file included from /opt/ros/rolling/include/rclcpp/client.hpp:40,
                 from /opt/ros/rolling/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/rolling/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/rolling/include/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/rolling/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/rolling/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/rolling/include/rclcpp/executor.hpp:35,
                 from /opt/ros/rolling/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/rolling/include/rclcpp/executors.hpp:21,
                 from /opt/ros/rolling/include/rclcpp/rclcpp.hpp:146,
                 from /home/alpha/Projects/ros-gst-bridge/gst_bridge/include/gst_bridge/gst_bridge.h:14,
                 from /home/alpha/Projects/ros-gst-bridge/gst_bridge/include/gst_bridge/rosaudiosrc.h:25,
                 from /home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosaudiosrc.cpp:34:
/home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosaudiosrc.cpp: In function ‘void rosaudiosrc_sub_cb(Rosaudiosrc*, audio_msgs::msg::Audio_<std::allocator<void> >::ConstSharedPtr)’:
/home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosaudiosrc.cpp:559:42: warning: format ‘%s’ expects argument of type ‘char*’, but argument 5 has type ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} [-Wformat=]
  559 |       RCLCPP_ERROR(ros_base_src->logger, "audio format changed during playback, encoding %s != %s",
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  560 |         gst_bridge::getRosEncoding(GST_AUDIO_INFO_FORMAT(&(src->audio_info))), msg->encoding.c_str()); // XXX account for the override
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   |
      |                                   std::string {aka std::__cxx11::basic_string<char>}
/home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosaudiosrc.cpp:559:91: note: format string is defined here
  559 |       RCLCPP_ERROR(ros_base_src->logger, "audio format changed during playback, encoding %s != %s",
      |                                                                                          ~^
      |                                                                                           |
      |                                                                                           char*
In file included from /opt/ros/rolling/include/rclcpp/client.hpp:40,
                 from /opt/ros/rolling/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/rolling/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/rolling/include/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/rolling/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/rolling/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/rolling/include/rclcpp/executor.hpp:35,
                 from /opt/ros/rolling/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/rolling/include/rclcpp/executors.hpp:21,
                 from /opt/ros/rolling/include/rclcpp/rclcpp.hpp:146,
                 from /home/alpha/Projects/ros-gst-bridge/gst_bridge/include/gst_bridge/gst_bridge.h:14,
                 from /home/alpha/Projects/ros-gst-bridge/gst_bridge/include/gst_bridge/rosimagesrc.h:25,
                 from /home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosimagesrc.cpp:34:
/home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosimagesrc.cpp: In function ‘void rosimagesrc_sub_cb(Rosimagesrc*, sensor_msgs::msg::Image_<std::allocator<void> >::ConstSharedPtr)’:
/home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosimagesrc.cpp:552:42: warning: format ‘%d’ expects argument of type ‘int’, but argument 5 has type ‘size_t’ {aka ‘long unsigned int’} [-Wformat=]
  552 |       RCLCPP_ERROR(ros_base_src->logger, "image format changed during playback, step %d != %d", src->step, msg->step/msg->width);
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~
      |                                                                                                      |
      |                                                                                                      size_t {aka long unsigned int}
/home/alpha/Projects/ros-gst-bridge/gst_bridge/src/rosimagesrc.cpp:552:87: note: format string is defined here
  552 |       RCLCPP_ERROR(ros_base_src->logger, "image format changed during playback, step %d != %d", src->step, msg->step/msg->width);
      |                                                                                      ~^
      |                                                                                       |
      |                                                                                       int
      |                                                                                      %ld
---
Finished <<< gst_bridge [42.9s]

Summary: 3 packages finished [51.8s]
  1 package had stderr output: gst_bridge

BrettRD, 2021-04-03:
Can you expand on this?
That Colcon output looks like it built successfully.
(Thanks for trying it against rolling! I'd love to hear if anything changed)

The warnings really do need cleaning up, would you like to submit a pull request?

gc-robotics, 2021-04-04:
Hey  Brett,

Happy Easter.

The warning was actually raised at the first colcon build. Repeating the command, it just completes successful - as compiling has already been done and no warning is raised.

And yes, I will take a look at it and will raise a PR.

Thanks man.
Cheers,
Gian.

closed 2021-04-05 by BrettRD