BrettRD/ros-gst-bridge #18: no audio/video being rendered on client (webrtc)

gc-robotics, 2021-04-06:
Hi Brett,

I am running this GST pipeline inside ROS - basically, using your pipeline code:

```
webrtcbin name=sendrecv bundle-policy=max-bundle stun-server=stun://stun.l.google.com:19302
 videotestsrc is-live=true pattern=ball ! videoconvert ! vp8enc deadline=1 ! rtpvp8pay ! queue ! application/x-rtp,media=video,encoding-name=VP8,payload=97 ! sendrecv.
 audiotestsrc volume=0.3 is-live=true wave=red-noise ! audioconvert ! queue ! opusenc ! rtpopuspay ! application/x-rtp, media=(string)audio, payload=(int)111, clock-rate=(int)48000, encoding-name=(string)OPUS, encoding-params=(string)2, minptime=(string)10, useinbandfec=(string)1 ! sendrecv. 
```

I know that the integration works properly, as I tried to use rosimagesrc/sink etc. with success (I have checked topics and all..)

The pipeline above seems to work, connects to the webrtc server, performs ice exchange, establish a connection with the other peer but, I am not able to see anything on the receiving client.

Could you advise on how I can debug this scenario? 
How can I check if the SDP packages are valid, etc.? 

Thanks in advance,
GC

BrettRD, 2021-04-12:
There are a few known bugs between what browsers do and what the standards say, I've run into a couple.

Your browser can display a history of the ice exchange, normally it'll tell you if negotiation fails but not always why.

gc-robotics, 2021-04-25:
how can I check if I hit one of those scenarios? Thanks

BrettRD, 2021-06-02:
I'm not a webrtc expert so I won't be much use debugging webrtc,  I only have a browser and a compiler.
Your browser should have the best idea of what's going on, you can look that up by typing in the urls `about:about` and `about:webrtc`