BrettRD/ros-gst-bridge #19: Feature request: rostextsrc element

clydemcqueen, 2021-04-11:
Brett, what do you think about adding a `rostextsrc` element to this package? Proposed usage:

~~~
gst-launch-1.0 -v rostextsrc topic="/string" ! txt. videotestsrc ! textoverlay name=txt ! autovideosink
~~~

I started in on this to understand how it might work. It compiles & is visible with gst-inspect-1.0, but it doesn't work yet. Let me know if you're interested in a PR. https://github.com/clydemcqueen/ros-gst-bridge/tree/clyde_rostext

Thanks,
/Clyde

BrettRD, 2021-04-12:
I've been thinking about this element for a while and I'd love to merge yours into the repo!

It's close enough to being viable as the opaque block that I wanted to try adding the `ANY` caps to it but I couldn't figure out how to do it nicely.

Make a pull request, it looks like you're only running into trouble at caps negotiation.
The sinks are a lot easier to write, and that'll add pocketsphinx to the list of turn-key features.

clydemcqueen, 2021-04-12:
Hey, Brett: I got negotiation working, but now I'm running into problems with timestamps. I submitted an early PR, we can shift the conversation there. Thanks!

BrettRD, 2021-06-02:
The text src is terrific as is, let's open some new issues to track usability problems

closed 2021-06-02 by BrettRD