BrettRD/ros-gst-bridge #27: include could not find load file:      /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/rosidl_cmake-extras.cmake

mutantbob, 2021-09-16:
While trying to build the `gst_bridge` I encountered the following error

```
~/vendor/ros-gst-bridge/gst_bridge/build$ ( . /opt/ros/galactic/setup.bash ; audio_msgs_DIR=../../audio_msgs/build/ament_cmake_core cmake ..)
-- Found ament_cmake: 1.1.4 (/opt/ros/galactic/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Found rclcpp: 9.1.0 (/opt/ros/galactic/share/rclcpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Found rosidl_adapter: 2.2.1 (/opt/ros/galactic/share/rosidl_adapter/cmake)
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 3.3.1 (/opt/ros/galactic/share/rmw_implementation_cmake/cmake)
-- Found rmw_cyclonedds_cpp: 0.22.3 (/opt/ros/galactic/share/rmw_cyclonedds_cpp/cmake)
-- Using RMW implementation 'rmw_cyclonedds_cpp' as default
-- Checking for module 'gstreamer-mpegts-1.0>=1.4.0'
--   No package 'gstreamer-mpegts-1.0' found
-- Checking for module 'gstreamer-codecparsers-1.0 >= 1.8.3'
--   No package 'gstreamer-codecparsers-1.0' found
-- Found audio_msgs: 0.0.0 (/home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core)
CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/rosidl_cmake-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/ament_cmake_export_dependencies-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/ament_cmake_export_libraries-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/ament_cmake_export_targets-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/ament_cmake_export_include_directories-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/rosidl_cmake_export_typesupport_libraries-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

CMake Error at /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/audio_msgsConfig.cmake:41 (include):
  include could not find load file:

    /home/thoth/vendor/ros-gst-bridge/audio_msgs/build/ament_cmake_core/rosidl_cmake_export_typesupport_targets-extras.cmake
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_package)

-- Found sensor_msgs: 2.2.3 (/opt/ros/galactic/share/sensor_msgs/cmake)
-- Configuring incomplete, errors occurred!
See also "/home/thoth/vendor/ros-gst-bridge/gst_bridge/build/CMakeFiles/CMakeOutput.log".
See also "/home/thoth/vendor/ros-gst-bridge/gst_bridge/build/CMakeFiles/CMakeError.log".
```

What mistake am I making?

BrettRD, 2021-09-17:
I'm not familiar with raw cmake builds of ROS2 packages.

It looks like it's called cmake for audio_msgs but hasn't passed the ros environment to it.\
You're also missing at least the system gstreamer install.

Dependencies are listed in `package.xml`, and get converted to apt package names at index.ros.org.

The easiest way to resolve this is to use a full ROS2 workspace, and run rosdep over it to call apt with the dependency lists in the various `package.xml` files.

```
# make a workspace, enter it, clone the repo into workspace/src
mkdir -p galactic_ws/src
cd galactic_ws
git clone https://github.com/BrettRD/ros-gst-bridge.git src/ros-gst-bridge

# initialise rosdep
sudo rosdep init
rosdep update

# let rosdep know what ros packages we already have
. /opt/ros/galactic/setup.bash 

# fetch remaining system dependencies for all packages in src/
rosdep install --from-paths src --ignore-src -r -y

# build everything in src/
colcon build

```

mutantbob, 2021-09-17:
Almost all of those steps were new to me (since I was only recently assigned to a ROS sprint).  I recommend you add them to the `README.md` to help folks get started, or include a link to a primer that provides an overview of the ROS2 workspace spellbook.

I bumped into a few hiccups following those instructions.
```
# rosdep init
ERROR: default sources list file already exists:
        /etc/ros/rosdep/sources.list.d/20-default.list
```

perhaps the ubuntu install of ros-galactic-desktop already did this?  https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html

```
~/vendor/galactic_ws$ colcon build
colcon: command not found
```

I'm going to go digging to see where that comes from.  Ubuntu usually suggests a package to install (like it did for `rosdep`), but didn't have a guess for how to get colcon.

mutantbob, 2021-09-17:
I was able to get colcon for ubuntu by following the instructions at https://colcon.readthedocs.io/en/released/user/installation.html

It appears to have built, although it took me a while to figure out the exact invocation of `gst-inspect`

``~/vendor/galactic_ws$ LD_LIBRARY_PATH=install/audio_msgs/lib:$LD_LIBRARY_PATH gst-inspect-1.0 ./install/gst_bridge/lib/gst_bridge/librosgstbridge.so
Plugin Details:
  Name                     rosgstbridge
  Description              ROS topic bridge elements
  Filename                 ./install/gst_bridge/lib/gst_bridge/librosgstbridge.so
  Version                  0.0.0
  License                  LGPL
  Source module            ros_gst_bridge
  Binary package           ros_gst_bridge
  Origin URL               https://github.com/BrettRD/ros-gst-bridge

  rosaudiosink: rosaudiosink
  rosimagesink: rosimagesink
  rostextsink: rostextsink
  rosaudiosrc: rosaudiosrc
  rosimagesrc: rosimagesrc
  rostextsrc: rostextsrc

  6 features:
  +-- 6 elements
```

BrettRD, 2021-09-18:
I made a commit to the readme after your first issue adding some short build notes, linking to colcon instructions and environment setup, can I get your opinion on that?

You can short-cut that LD preload step by sourcing `install/setup.bash` from the workspace.

`opt/ros/galactic/setup.bash` adds ros objects, yaml parsers etc from the desktop install to your environment,
`install/setup.bash` adds anything built by your workspace to your environment.

I've built this from and for a ROS2 developers' perspective, so I really appreciate the GStreamer dev perspective.

mutantbob, 2021-09-21:
The changes to `README.md` are a good start.  I'm trying to build a `Dockerfile` that builds this git repo from a baseline ubuntu install (which is a gold standard of build instructions), but so far I'm missing something from the environment or build instructions and it does not generate a `.so` file.

```
#
# The purpose of this Dockerfile is to demonstrate the build
# environment and procedure needed to successfully compile the
# ros-gst-bridge package
#

FROM ubuntu:focal as ros_base

RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get -y install \
    apt-utils \
    curl \
    git \
    gnupg2 \
    locales \
    lsb-release

#
# https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html
#

RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | tee /etc/apt/sources.list.d/ros2.list

# your locale might already be configured this way on an interactive (non-docker) system
RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8

# this will result in fetch of >1650 packages in this small docker environment
RUN apt-get update && DEBIAN_FRONTEND="noninteractive" apt-get -y install ros-galactic-desktop python3-rosdep python3-colcon-core
RUN bash -c '. /opt/ros/galactic/setup.bash && \
    rosdep init && \
    rosdep update '

#
#
#

FROM ros_base

RUN mkdir -p /galactic_ws/src
WORKDIR /galactic_ws/src
RUN git clone https://github.com/BrettRD/ros-gst-bridge.git 

RUN bash -c 'source /opt/ros/galactic/setup.bash && \
    DEBIAN_FRONTEND=noninteractive rosdep install --from-paths . --ignore-src -r -y'

RUN bash -c 'source /opt/ros/galactic/setup.bash && \
    colcon build '
```

If you want to experiment with docker, save the above as a `Dockerfile` and run `docker build -t ros-gst-bridge .` (assuming `.` is the directory containing the `Dockerfile`)

mutantbob, 2021-09-21:
My `Dockerfile` was missing `python3-colcon-common-extensions`.  I have expanded the BUILD section of your `README.md` and created a pull request for you to review.

BrettRD, 2021-09-22:
Thank you very much!

closed 2021-09-23 by BrettRD