BrettRD/ros-gst-bridge #30: CI improvements

BrettRD, 2021-09-22:
Basic CI is up, it confirms the code builds, but nothing else.

- [ ] Create a testing package
- [ ] Build with gst_bridge -werror
- [ ] Test gst-inspect-1.0 can load the plugin
- [ ] Basic functional testing
- [ ] Unit testing
- [ ] Code coverage
- [ ] Fuzzing with [hypothesis](hypothesis.readthedocs.io)

BrettRD, 2022-06-17:
Thanks for the check, I don't think CI has run in a while
This checklist is for automated testing through github runners, I'd like to maintain a colcon test command that confirms the pipeline safely enters playback and moves a few image frames

sandman, 2023-05-23:
I've been meaning to do something similar for a while now. I think there are broadly two approaches to functional testing of this ROS-Gstreamer pipeline:
1. The ROS way: (your suggestion) using colcon test to wrap 
2. The GStreamer way:  use gst-validate or a similar pipeline-level test framework and communicate results with a CI tool like Jenkins. 

I'll have a dig at this in the coming weeks.

BrettRD, 2023-05-25:
I'm aiming for a major release from the feat_modular branch very soon, it has a much more sensible pipeline handler and a basic colcon test configuration you can work with