BrettRD/ros-gst-bridge #33: How can you make rosimagesrc generate an End Of Stream event to eventually end the pipeline?

mutantbob, 2021-09-28:
I am experimenting with the `rosimagesrc`.  I am playing a bag file that someone else recorded.  I am using the pipeline 
```
gst-launch-1.0 rosimagesrc ros-topic=/cam0/image_raw ! videoconvert ! x264enc ! mpegtsmux ! filesink location=/tmp/x.ts
```
I would use `mp4mux ! filesink location=/tmp/x.mp4`, but MP4 files aren't valid until you have ended the capture process and finalized the atoms*.  But I do not know how to make the `rosimagesrc` end.  For now I am faking it by hitting `^C` until I kill gst-launch (which is why I have to use a streaming format like mpegts).

*yes, there is a way to create a streaming mp4 using `moof`s , but I do not think `mp4mux` has code to implement that

BrettRD, 2021-09-29:
That is an excellent question.

In your specific case, [`splitmuxsink`](https://gstreamer.freedesktop.org/documentation/multifile/splitmuxsink.html) might have the tools to make a workaround

The best option is probably to extend rosbasesrc with a subscription that triggers gstreamer events like eos.
Then the EOS can be triggered by a message tacked onto the end of the bag, (or played from a second bag with one message)
This would also create appropriate entry-points to manage seek events and latency later on

I can make the ROS API up to the `gst_pad_send_event()`, but I'm not sure how to actually do EOS correctly.

potentially relevant: https://github.com/BrettRD/ros-gst-bridge/issues/12

@KibaekJeong How did you do EOS and closing files in your work? (does github make a notification?)

gowthamhosamane, 2022-01-21:
It's very useful and thanks for the great work @BrettRD 

Was just curious about this issue, one way to trigger EOS is to timeout (may be 5s or 10s or 60s!) while waiting for the message. Additionally, GstBaseSrc implements events function but if overridden inside rosimagesrc, that function can just reset the subscription (like in close()) and call default event handler. This way if -e option is used in gst-launch application, it takes care of cleaner exit as well as proper closing of MP4 files or similar requirements.