BrettRD/ros-gst-bridge #46: ROS1 Port

JonasHablitzel, 2022-03-09:
Hello @BrettRD,

I'm trying to port this package to ROS1 (Melodic) and wanted to ask you first 
if this is ok and if you have any plans in mind on how to do it?
I looked a little bit around and found two possible approaches.

**1.** The first one would be that another branch for ros1 will be created.
This would be the easiest but would make it harder to support both Versions

**2.** Another option would be similar to ros_deep_learning (https://github.com/dusty-nv/ros_deep_learning/)
There they created some types and functions dynamically, depending on the Version ([ros_compat.h](https://github.com/dusty-nv/ros_deep_learning/blob/60f846653bb653b95512df728e34b767f733e99f/src/ros_compat.h))

I already tried to implement the second Option and got it working for the Message. But I got stuck on the rclcpp::Context and rclcpp::Executor in the Base Classes. Because there are no equivalent constructs in ROS1. Do you maybe have some ideas to get around this? and what would be the preferred option for you?

Thank you in advance.

BrettRD, 2022-03-10:
Hi @JonasHablitzel,

A port would be great, there's no technical reason it wouldn't work in ROS1 aside from the transport overheads

I already have a branch for Dashing, so I'm stuck with a git cherry-pick workflow anyway.
I don't think I'd enjoy using a macro-heavy shim; ros-gst-bridge breaks a couple of fundamental ROS conventions, and it sounds like you've already discovered those.

ROS1 was built for one-node-one-process, and you have to work pretty hard to break that assumption even in ROS2.

The Context allows each GStreamer element to interact with different ROS DDS domains
The Executor hosts the node's callback-groups in (or near) the thread that GStreamer provides for the element to live in.  Without the separate Executor, every node in the pipeline would share one thread for their callbacks, regardless of how many threads GStreamer used.

You'll have to do some digging into how you can isolate your callback queues, node handles, and node name; and how to start the node without launch. That's definitely the tricky part.

Start a new branch, let me know which commit you forked from and I'll make space for it here.

JonasHablitzel, 2022-03-10:
Hi @BrettRD,

thanks for your advice! I will look into these topics.
I forked it from the most recent **ros2** and named it **melodic-devel**.

scottrfrancis, 2023-01-26:
@JonasHablitzel - did you ever complete the port?  I have interest in a Noetic version and was hoping I might be able to work from your Melodic version.

jobafr, 2023-03-09:
Hi @JonasHablitzel, @scottrfrancis and folks landing here from their favorite search engine,

I have written a ROS1←→GStreamer plugin that is now pending [as a merge request in the gst-plugins-rs repo](https://gitlab.freedesktop.org/gstreamer/gst-plugins-rs/-/merge_requests/1126). It's written in Rust and therefore not based on @BrettRD 's code here, but should serve the same purpose. However, only image topics are supported as of now.

I'd appreciate if anyone would test that code and provide feedback (here or on the GStreamer gitlab).

zhh2005757, 2024-03-20:
> Hi @JonasHablitzel, @scottrfrancis and folks landing here from their favorite search engine,
>
> I have written a ROS1←→GStreamer plugin that is now pending [as a merge request in the gst-plugins-rs [...]

why cannot build? 
"gst-plugin-rosimage` not found in workspace `/home/nvidia/gst-plugins-rs`), exit_code: 101 "

zhh2005757, 2024-03-20:
> Hi @JonasHablitzel, @scottrfrancis and folks landing here from their favorite search engine,
>
> I have written a ROS1←→GStreamer plugin that is now pending [as a merge request in the gst-plugins-rs [...]

I have encountered such a problem when building your package

error[E0603]: module `traits` is private
--> video/rosimage/src/rosimagesrc/imp.rs:6:15
   |
6  | use gst_base::traits::BaseSrcExt;
   |               ^^^^^^ private module
   |
note: the module `traits` is defined here
  --> /home/nvidia/.cargo/git/checkouts/gstreamer-rs-79e52a2d27eb91a3/983e8b3/gstreamer-base/src/lib.rs:24:9
   |
24 | pub use crate::auto::*;
   |         ^^^^^^^^^^^

My platform is nvidia xavier.  What should I do?