BrettRD/ros-gst-bridge #48: Feature request: Multiple outputs with synchronized timestamps

Rubiaceae65, 2022-07-12:
I am trying to publish stereo images from a playstation 5 camera, but to do stereo processing properly timestamps should be synchronized, it seems that with this pipeline they are not: 

<code>
gst-launch-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ v4l2src extra-controls="c,exposure_auto=0,power_line_frequency=1" ! video/x-raw, height=1080, width=3840 \
        ! videoconvert  ! queue !  tee name=tee \
        ! queue !  videocrop bottom=0 top=0 left=1920 right=0 !   rosimagesink ros-name="ps5eye_left" ros-frame-id="ps5eye_left" ros-namespace="ps5eye" ros-topic="left/image_raw" \
        tee. ! queue ! videocrop right=1920 ! rosimagesink ros-name="ps5eye_right" ros-frame-id="ps5eye_right" ros-namespace="ps5eye" ros-topic="right/image_raw"
</code>

BrettRD, 2022-07-13:
This is caused by https://github.com/BrettRD/ros-gst-bridge/issues/37, the timestamps differ on the order of microseconds because of a hacky clock sampling step at startup.

a gstreamer bus message to coordinate ros-time with pipeline time would solve this permanently, but I haven't explored the full implications of that. 

As a workaround, take a look at the `ros-start-time` element parameter, this will allow you to specify the time offsets explicitly on each ros element, bypassing the buggy clock sampling step

Rubiaceae65, 2022-07-13:
I tried that, but the disparity node still wasn't outputting anything, when i give it the option approximate_sync, it works.

Haven't properly looked at the timestamps yet, so not sure why it didn't work with passing ros-start-time.