BrettRD/ros-gst-bridge #54: gst pipeline - rosimagesrc with udpsink

esharet, 2023-01-31:
Hi,
I am trying to launch gstreamer pipeline of rosimagesrc from gazebo to udpsink and I have a very big latency (~6 sec). I am wondering if my pipeline is right:
```
Gst.parse_launch("""rosimagesrc ros-topic=/camera_right/image_raw ! videoconvert ! x264enc ! video/x-264,stream-format=byte-stream ! rtph264pay config-interval=1 pt=96 ! udpsink host=127.0.0.1 port=5000 sync=false""")
```

when openning the stream I am using:
```
gst-launch-1.0 udpsrc port=5000 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink fps-update-interval=1000 sync=false
```

Also sometime I have a critical error: Gstreamer-CRITICAL **:04:47:37.658: gst_buffer_map_range: assertion 'GST_IS_BUFFER (buffer)' failed
Maybe I an using the plugins wrong.
Thank you,
Ehud

esharet, 2023-02-26:
Solved the problem!
Gst.parse_launch("""rosimagesrc ros-topic=/camera_right/image_raw ! queue max-size-buffers=1 ! videoconvert ! x264enc ! video/x-264,stream-format=byte-stream ! rtph264pay config-interval=1 pt=96 ! udpsink host=127.0.0.1 port=5000 sync=false""")

closed 2023-03-24 by BrettRD

djaniel, 2024-10-31:
Hello, 
I am also trying to stream an image using your gstreamer plugin. When I try your pipeline, I get:
```
WARNING: erroneous pipeline: could not link x264enc0 to rtph264pay0, neither element can handle caps video/x-264, stream-format=(string)byte-stream
```
However, there are some pipelines that are working. The pipeline I used to display the video, although a bit laggy, it works:

```
rosimagesrc ros-topic=/camera/image_raw ! videoconvert ! autovideosink
```

If I use `gst-launch` in verbose mode, I see that all properties on the pads are set correctly. 

My best tries for sending video using UDP are, for the server:
```
gst-launch-1.0 --gst-plugin-path=./install/gst_bridge/lib/gst_bridge/ rosimagesrc ros-topic=/camera/image_raw ! videoconvert ! x264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5000
```
and for the client
```
gst-launch-1.0 -e -v udpsrc port=5000 ! application/x-rtp, encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! autovideosink
```
But then I get a lot of warnings on the client and no video:
```
WARNING: from element /GstPipeline:pipeline0/GstRtpJPEGDepay:rtpjpegdepay0: Empty Payload.
Additional debug info:
../gst/rtp/gstrtpjpegdepay.c(738): gst_rtp_jpeg_depay_process (): /GstPipeline:pipeline0/GstRtpJPEGDepay:rtpjpegdepay0
```
I'd greatly appreciate any suggestions,
Thanks

BrettRD, 2024-10-31:
The Gstreamer community IRC and Discourse are the best place for pure pipeline questions, and it's impolite to hijack a github issue
Queues will often improve your lag, and your udp pipelines are trying to decode h264 as jpeg