BrettRD/ros-gst-bridge #56: Not able to launch pipeline_node

sarithabapuram, 2023-03-23:
Hi,

Not able to launch the gst pipeline node with launch file, could you please share some example for the same.

~/workspaces/gstreamer_ws$ ros2 launch gst_bridge example.launch.py
[INFO] [launch]: All log files can be found below /home/avlabs/.ros/log/2023-03-24-05-21-48-934184-nuvo-one-234474
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [pipeline_node-1]: process started with pid [234475]
[pipeline_node-1] [DEBUG] [1679615509.086904536] [rcl]: Initializing node 'pipeline_node' in namespace ''
[pipeline_node-1] [DEBUG] [1679615509.086944619] [rcl]: Using domain ID of '0'
[ERROR] [pipeline_node-1]: process has died [pid 234475, exit code -11, cmd '/home/avlabs/workspaces/gstreamer_ws/install/gst_pipeline/lib/gst_pipeline/pipeline_node --ros-args --log-level debug --ros-args -r __node:=gst_pipeline_node --params-file /home/avlabs/workspaces/gstreamer_ws/install/gst_bridge/share/gst_bridge/config/example.config.yaml'].

Thanks in Advance

BrettRD, 2023-03-24:
Your logs look normal right up until the error.
-11 looks like a segfault, which CPU are you using?

sarithabapuram, 2023-03-24:
Intel(R) Xeon(R) E-2278GE CPU

And I kept only audio_test as part of simple_bins, and removed rest others including webrtc_example_bin section

BrettRD, 2023-03-24:
That does narrow it down a lot.
The only code that example uses is the python pipeline host, it sounds like your gstreamer is having a bad time

Does the audio_test pipeline work on its own through gst-launch?

```
$ gst-launch-1.0 audiotestsrc volume=0.3 is-live=true wave=red-noise ! queue ! audioconvert ! alsasink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
```

arukan404, 2023-04-30:
I am trying to run the example launch pipeline and running into the following type error issue
[pipeline_node-1] rclpy.exceptions.InvalidParameterTypeException: Trying to set parameter 'gst_plugins_required' to '['rosgstbridge', 'videotestsrc', 'audiotestsrc', 'alsa', 'webrtc', 'nice', 'opus', 'vpx', 'dtls', 'srtp', 'rtp', 'rtpmanager']' of type 'STRING_ARRAY', expecting type 'BYTE_ARRAY'

I saw that there's a mention of this issue in the following lines in pipeline_node.py, lines 27-33
# XXX this is an awful hack to deal with broken param type mappings in galactic.
# https://github.com/ros2/rclpy/issues/829#issuecomment-937517881
rosdistro=os.environ['ROS_DISTRO']
if rosdistro == 'galactic' or rosdistro == 'humble':
  distro_dynamic_typing={'dynamic_typing':True}
else:
  distro_dynamic_typing={}
  
 I am testing this in humble