BrettRD/ros-gst-bridge #57: gst pipeline node as a composable node

rawnam, 2023-10-02:
Hi,

I was wondering if it would be possible to run the pipeline node as a composable node which can then be composed within a multi-threaded executor. Would it be possible to do so? How could we go about it? Will it be possible to move the pointers from the ros-image-sink element to any downstream application? 

Do we need to compose the pipeline node for this?

Would love to hear some thoughts on this.

Thanks

BrettRD, 2023-10-03:
Hi,

Take a look at the [feature/param_plugin](https://github.com/BrettRD/ros-gst-bridge/tree/feature/param_plugin) branch

Node composition wasn't possible under rclpy, so I re-wrote the pipeline handler in C++, and added tools to pass external node-interfaces into the bridge plugins, so that the bridge plugins create their publishers and subscribers on the pipeline-node's node interfaces.

In theory, this allows the bridge elements to operate as a composable node by piggybacking on the new C++ pipeline-node. I don't yet have tests for it, so there's a chance the multi-threaded executor might break it.

I'm expecting to properly release this feature branch by the end of the year and fully deprecate the python node, I'd love to hear if it works for you or what breaks when you try it.

Cheers

BrettRD, 2024-03-17:
update: this is now on the `develop` branch, fully functional, but still waiting on a proper shake-down

closed 2025-02-07 by BrettRD