BrettRD/ros-gst-bridge #64: Segfault on startup

fbe555, 2024-03-15:
I am trying to use the ros-gst-bridge package for a ROS2 project, and I cannot currently get it to work, as it exits with a segfault right away. I have performed the following steps:

- Setup a docker based ros2 environment as per [this](https://docs.ros.org/en/rolling/How-To-Guides/Setup-ROS-2-with-VSCode-and-Docker-Container.html) page, using iron as my ros distro.
- clone the ros2 branch of this repo into the ros2 workspace
- run ```source /opt/ros/iron/setup.bash```
- run ```sudo rosdep update && rosdep install -yir --from-paths .```
- try ```colcon build```

At this point it seems like some packages are still missing, so I ran the following commands to install what appeared to be missing:
```bash
  sudo apt install ros-iron-diagnostic-updater
  sudo apt install python3-cairo
  sudo -H pip install asyncio-gst
```
At this point I then continue:
- run ```colcon build``` successfully
- run ```source install/setup.bash```
I then run the example launch with the following result:
```bash
felix@Felix-T490s-U2210:/home/ws$ ros2 launch gst_bridge example.launch.py 
[INFO] [launch]: All log files can be found below /home/felix/.ros/log/2024-03-15-12-08-12-819640-Felix-T490s-U2210-17249
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [pipeline_node-1]: process started with pid [17256]
[pipeline_node-1] [DEBUG] [1710504493.081854954] [rcl]: Initializing node 'pipeline_node' in namespace ''
[pipeline_node-1] [DEBUG] [1710504493.081879735] [rcl]: Using domain ID of '42'
[ERROR] [pipeline_node-1]: process has died [pid 17256, exit code -11, cmd '/home/ws/install/gst_pipeline/lib/gst_pipeline/pipeline_node --ros-args --log-level debug --ros-args -r __node:=gst_pipeline_node --params-file /home/ws/install/gst_bridge/share/gst_bridge/config/example.config.yaml'].
```

This does not only happen for the example package, but also when I create my own launch with a simple pipeline.
I have edited the example config so that only the audiotest remained, and replaced the alsasink (which i'm unsure about when inside a container) and replaced it with a fakesink. I then ran that pipeline using gst-launch-1.0 to verify it, and that works. The result when using the gst_pipeline node is always as above. 

Let me know if I forgot any important details.

Thank you for your time and work
Felix

BrettRD, 2024-03-17:
Thanks for the breakdown, I've not yet tried out ROS Iron

If you're not using any ros elements in the pipeline, then it sounds like this is an issue in the pipeline_node.
The python version of the pipeline_node uses some unmaintained glib compatibility tools, and this might be the first signs they won't survive upstream.

Could you try out the much more maintainable C++ pipeline_node in the `develop` branch?

fbe555, 2024-03-17:
Switching to the devel branch and following the same procedure, but without the additional package installs, I get the following build error:
```bash
...
Starting >>> gst_pipeline_plugins_webrtc                 
--- stderr: gst_pipeline_plugins_webrtc                                                                               
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:43 (pkg_check_modules)

---
Failed   <<< gst_pipeline_plugins_webrtc [4.66s, exited with code 1]
Aborted  <<< gst_pipeline_plugins [5.30s]         
```
If i ignore the package, everything else seems to work fine. I get an authorization error when I run the rx example, but it works if I replace the ximagesink with an autovideosink, so I suspect it's something with the container configuration regarding displays. 

Let me know if I can help with the webrtc plugin, however I don't need it, so I'm happy as is.

BrettRD, 2024-03-18:
gst_pipeline_plugins_webrtc pkg_check_modules on [line 43](https://github.com/BrettRD/ros-gst-bridge/blob/develop/gst_pipeline_plugins_webrtc/CMakeLists.txt#L43) looks like the libsoup dependency
Ideally that would come across through rosdep from its listing on [ros index](https://index.ros.org/search/?term=libsoup&section=deps) but it hasn't been added.
Do you know how to add system dependencies to the index?

Glad to hear you have a path forward

fbe555, 2024-03-18:
I made a pull request to ros index ([here](https://github.com/ros/rosdistro/pull/40241)) and one for develop in this repo ([here](https://github.com/BrettRD/ros-gst-bridge/pull/65)).
I haven't updated ros index before, but I've followed the guidelines and tested that it works for me. Let me know if it ok.

fbe555, 2024-03-29:
I made another PR to rosindex for the lib-soup dependency, and will make a PR here for the final changes needed once the rosindex one gets merged.   

My project now runs, but with one caveat. The pipeline I'm using only works when run through `gst-launch-1.0`, and not when run in a _gst_pipeline_ node. I have tested that the _gst_pipeline_ node works by using it to create a pipeline identical to the gamma correction example from the README, which works as intended, however the following node:
```python
    Node(
        name='gst_rover_video_capture',
        package='gst_pipeline',
        executable='gst_pipeline_node',
        arguments=['--ros-args', '--log-level', 'debug'],
        parameters=[config_path],
    )
``` 
with the following config file:
```yaml
  gst_rover_video_capture:
    ros__parameters:

      gst_plugin_paths:
        - 'install/gst_bridge/lib/gst_bridge/'

      pipeline_base_descr: udpsrc address=1.0.0.0 port=5600 ! "application/x-rtp" ! rtph264depay ! decodebin ! queue ! videoconvert ! rosimagesink ros-topic="rover_camera" sync=false
```
doesn't publish anything, while: 
```bash
    #!/usr/bin/bash 
    gst-launch-1.0 --gst-plugin-path=$1  udpsrc address=0.0.0.0 port=5600 ! "application/x-rtp" ! rtph264depay ! decodebin ! queue ! videoconvert ! rosimagesink ros-topic="rover_camera" sync=false
```
Works fine.   

I'm unsure how best to debug this though, as I'm new to gstreamer. On the commandline I have used `fakesink dump=1` to test if anything is getting through to a specific point in a pipeline, but I'm unsure how to do a similar thing in the _gst_pipeline_ node.

BrettRD, 2024-03-29:
Any chance it's the different udpsrc IP addresses?

I vaguely remember being able to use the gst debug environment variables ahead of the ros2 launch command the same way as they are used ahead of gst-launch. I don't remember the syntax though.

You can also use the buffer_observer plugin to get a ros message with a timestamp whenever a gst buffer is produced from a given element.
The gui sink elements like ximagesink should work too, and you can use `tee` and leaky queues to run both sinks simultaneously.
Have you tried a simple `videotestsrc ! videoconvert ! queue ! ximagesink` through the pipeline node?

fbe555, 2024-03-29:
Don't know where the '1.' came from in the above, but it wasn't in my working directory, and I just rechecked that it doesn't work 🤷 I also tested videotestsrc -> ximagesink as described, which works fine. I will use a bash script as a node for now, and I'll see if I can find some time to look into it further when the other merge requests come through.