BrettRD/ros-gst-bridge #71: ros topic to gstream

jprodriguezg, 2024-12-05:
Hello there,

I am trying to run a simple example that converts a ros2 topic into a GStreamer and shows it in a window. I am getting the images from the topic `/cam0/image_raw_color` and launching the following to do the conversion:

```
gst-launch-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ rosimagesrc ros-topic=/cam0/image_raw_color !  queue ! video/x-raw, format=BGRA,width=640,height=360,framereate=30/1 ! xvimagesin
```
I got the format, width, height, and framerate info from the topic information.  I am getting the following error
```
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
[INFO] [1733393414.390777065] [gst_image_src_node]: getcaps with filter 'NULL'
[INFO] [1733393414.390824962] [gst_image_src_node]: waiting for first message
[INFO] [1733393414.390783829] [gst_image_src_node]: stream_start at 1733393414390783157
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/Rosimagesrc:rosimagesrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/Rosimagesrc:rosimagesrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.073234998
Setting pipeline to NULL ...
Freeing pipeline ..
```

Do you have any ideas on how to solve this problem? I can see the images from the topic using `rqt` successfully.

BrettRD, 2024-12-08:
Without digging deeper, I'd point to the frame rate or the format.
rosimagesrc can't do any image processing at all and ros has no idea what framerate image topics run at, so specifying caps is asking for negotiation errors.
A videoconvert element is cheap on performance if it's just shuffling bytes, so use that instead of trying to force formats.

closed 2024-12-08 by BrettRD

jprodriguezg, 2024-12-09:
Hi @BrettRD thanks for the feedback

I confirm that using `videoconvert` solves the compatibility issue
```
gst-launch-1.0 --gst-plugin-path=install/gst_bridge/lib/gst_bridge/ rosimagesrc ros-topic=/cam0/image_raw_color !  queue ! video/x-raw ! videoconvert ! autovideosink
```