BrettRD/ros-gst-bridge #75: Stream stops working after a few seconds

breqdev, 2025-01-24:
Hi there!

I'm having an issue using `rosimagesrc` with the `usb_cam` package. Right now, my setup includes:

- An [Arducam B0292](https://www.amazon.com/Arducam-Autofocus-Microphone-Computer-Raspberry/dp/B08RHTG845) camera
- The `usb_cam_node_exe` node running, outputting `/usb_cam/image_raw` and `/usb_cam/image_raw/compressed` (among other topics)
- A GStreamer pipeline using `rosimagesrc` to pull images from `usb_cam_node_exe`.
- `RMW_IMPLEMENTATION` set to `rmw_cyclonedds_cpp`, and all nodes running on the same host

I'm able to run `rqt_image_view` and subscribe to `/usb_cam/image_raw/compressed` to view the camera stream continuously, and I'm also able to `ros2 topic echo /usb_cam/image_raw` to validate that image data is being published continuously on that topic. I get the expected framerate of around 30 fps.

However, when I try to run the example pipeline (or any pipeline with `rosimagesrc`) on this input, the pipeline works normally for a few seconds and then stalls:

<details>
<summary>Full command output</summary>

```
rover@base-station:~/rover/ros2_ws (main) $ gst-launch-1.0 --gst-plugin-path=/home/rover/rover/submodule_ws/install/gst_bridge/lib/gst_bridge -v rosimagesrc ros-topic="/usb_cam/image_
raw" ! gamma gamma=2.0 ! rosimagesink ros-topic="image_gamma_corrected"
Setting pipeline to PAUSED ...
[WARN] [1737742017.110227812] [rclcpp]: logging was initialized more than once
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
[INFO] [1737742017.113722520] [gst_image_src_node]: getcaps with filter 'NULL'
[INFO] [1737742017.113760172] [gst_image_src_node]: waiting for first message
[INFO] [1737742017.113740118] [gst_image_src_node]: stream_start at 1737742017113739577
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:22.872195157
Setting pipeline to NULL ...
^C
rover@base-station:~/rover/ros2_ws (main) $ gst-launch-1.0 --gst-plugin-path=/home/rover/rover/submodule_ws/install/gst_bridge/lib/gst_bridge -v rosimagesrc ros-topic="/usb_cam/image_raw" ! gamma gamma=2.0 ! rosimagesink ros-topic="image_gamma_corrected"
Setting pipeline to PAUSED ...
[WARN] [1737742041.009261417] [rclcpp]: logging was initialized more than once
Pipeline is live and does not need PREROLL ...
[INFO] [1737742041.013471393] [gst_image_src_node]: getcaps with filter 'NULL'
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
[INFO] [1737742041.013504321] [gst_image_src_node]: waiting for first message
[INFO] [1737742041.013539688] [gst_image_src_node]: stream_start at 1737742041013538869
New clock: GstSystemClock
[INFO] [1737742041.022241014] [gst_image_src_node]: preparing video with caps 'video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1'
[INFO] [1737742041.022349229] [gst_image_sink_node]: preparing video with caps 'video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1'
[INFO] [1737742041.022364850] [gst_image_sink_node]: setcaps format string is RGB 
[INFO] [1737742041.022370069] [gst_image_sink_node]: setcaps n_components is 3
[INFO] [1737742041.022374342] [gst_image_sink_node]: setcaps bits is 8
[INFO] [1737742041.022378514] [gst_image_sink_node]: setcaps pixel_stride is 3
/GstPipeline:pipeline0/Rosimagesrc:rosimagesrc0.GstPad:src: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstGamma:gamma0.GstPad:src: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/Rosimagesink:rosimagesink0.GstPad:sink: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstGamma:gamma0.GstPad:sink: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
Redistribute latency...
WARNING: from element /GstPipeline:pipeline0/Rosimagesink:rosimagesink0: Pipeline construction is invalid, please add queues.
Additional debug info:
../libs/gst/base/gstbasesink.c(1249): gst_base_sink_query_latency (): /GstPipeline:pipeline0/Rosimagesink:rosimagesink0:
Not enough buffering available for  the processing deadline of 0:00:00.020000000, add enough queues to buffer  0:00:00.020000000 additional data. Shortening processing latency to 0:00:00.000000000.
WARNING: from element /GstPipeline:pipeline0/Rosimagesink:rosimagesink0: Pipeline construction is invalid, please add queues.
Additional debug info:
../libs/gst/base/gstbasesink.c(1249): gst_base_sink_query_latency (): /GstPipeline:pipeline0/Rosimagesink:rosimagesink0:
Not enough buffering available for  the processing deadline of 0:00:00.020000000, add enough queues to buffer  0:00:00.020000000 additional data. Shortening processing latency to 0:00:00.000000000.
[INFO] [1737742041.023934638] [gst_image_sink_node]: stream_start at 1737742041023931735
0:00:01.5 / 99:99:99.
```
</details>

Since a few of the error messages point to `queue` elements being needed, I tried adding those to the pipeline. This removed those errors but the pipeline still stopped working after a few seconds:

```
rover@base-station:~/rover/ros2_ws (main) $ gst-launch-1.0 --gst-plugin-path=/home/rover/rover/submodule_ws/install/gst_bridge/lib/gst_bridge -v rosimagesrc ros-topic="/usb_cam/image_raw" ! queue ! gamma gamma=2.0 ! queue ! rosimagesink ros-topic="image_gamma_corrected"
Setting pipeline to PAUSED ...
[WARN] [1737742649.164740571] [rclcpp]: logging was initialized more than once
Pipeline is live and does not need PREROLL ...
[INFO] [1737742649.168812727] [gst_image_src_node]: getcaps with filter 'NULL'
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
[INFO] [1737742649.168850048] [gst_image_src_node]: waiting for first message
[INFO] [1737742649.168911356] [gst_image_src_node]: stream_start at 1737742649168910494
New clock: GstSystemClock
[INFO] [1737742649.173977409] [gst_image_src_node]: preparing video with caps 'video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1'
/GstPipeline:pipeline0/Rosimagesrc:rosimagesrc0.GstPad:src: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:sink: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:src: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstGamma:gamma0.GstPad:src: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstQueue:queue1.GstPad:sink: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstGamma:gamma0.GstPad:sink: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstQueue:queue1.GstPad:src: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
[INFO] [1737742649.174237272] [gst_image_sink_node]: preparing video with caps 'video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1'
[INFO] [1737742649.174275157] [gst_image_sink_node]: setcaps format string is RGB 
[INFO] [1737742649.174292274] [gst_image_sink_node]: setcaps n_components is 3
[INFO] [1737742649.174297083] [gst_image_sink_node]: setcaps bits is 8
[INFO] [1737742649.174301898] [gst_image_sink_node]: setcaps pixel_stride is 3
/GstPipeline:pipeline0/Rosimagesink:rosimagesink0.GstPad:sink: caps = video/x-raw, format=(string)RGB, height=(int)480, width=(int)640, framerate=(fraction)0/1
Redistribute latency...
[INFO] [1737742649.176075218] [gst_image_sink_node]: stream_start at 1737742649176073268
0:00:08.6 / 99:99:99.
```

After the pipeline stops, hitting Ctrl+C does not always stop GStreamer and I need to press Ctrl+\ twice in order to exit.

```
^\Caught SIGQUIT
Spinning.  Please run 'gdb gst-launch-1.0 81547' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^\Quit (core dumped)
```

I also tried running with FastDDS instead of CycloneDDS and saw a different failure mode, in which I receive an image only about every 10-20 seconds or so (much lower than the consistent ~30fps with `rqt_image_view`). I don't believe this is due to slow processing -- I'm streaming at only 480p on a 13th gen i9 laptop.

Any thoughts as to why this might be happening or how I could debug the issue further?

BrettRD, 2025-02-07:
I've seen a lot of intermittent bugs in the DDS that make whole topics disappear and transmission to become unreliable.
If this is a DDS performance issue, then you want linux perf tools and a flame graph.
if it's DDS reliability, then you'll need to get familiar with restarting the dds daemon and exploring QoS settings on command line.

For the sake of ruling out the obvious, run these three streams in separate terminals.

`gst-launch-1.0 -v videotestsrc ! queue ! rosimagesink ros-topic="/image_raw"`

`gst-launch-1.0 -v rosimagesrc ros-topic="/image_raw" ! queue ! gamma gamma=2.0 ! rosimagesink ros-topic="/image_gamma_corrected"`

`gst-launch-1.0 -v rosimagesrc ros-topic="/image_gamma_corrected" ! queue ! videoconvert ! xvimagesink`

breqdev, 2025-03-13:
Yep, running those three seems to work as intended!

Based on your suggestion that it's a DDS problem, I followed the [Cyclone DDS Tuning](https://docs.ros.org/en/humble/How-To-Guides/DDS-tuning.html#cyclone-dds-tuning) section in the ROS docs to increase the socket receive buffer size to 10 MB and I get a different behavior now.

TL;DR: For some reason, everything works fine with a topic published by `rosimagesink` but problems start happening with a topic published by `usb_cam`.

I can run the `usb_cam` node and publish on `/usb_cam/image_raw`. From within `rqt_image_view`, the topic looks fine (consistent 30 FPS). But if I try to run a similar pipeline with `rosimagesrc`:

```
rosimagesrc ros-topic=/usb_cam/image_raw ! queue ! videoconvert ! xvimagesink
```

I usually get a frame only about twice a second, and the console is filled with:

```
WARNING: from element /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0: A lot of buffers are being dropped.
Additional debug info:
../libs/gst/base/gstbasesink.c(3143): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0:
There may be a timestamping problem, or this computer is too slow.
```

I don't think it's a CPU issue (the highest-usage CPU core in my system is only at around 15% utilization).

Here's a table to summarize what's going on:
- In one terminal I'm running the node which publishes the ROS image topic (`usb_cam` or `rosimagesink`).
  - The `rosimagesink` pipeline I'm running is `videotestsrc ! video/x-raw,width=640,height=480,framerate=30/1,format=RGB ! queue ! rosimagesink ros-topic=/usb_cam/image_raw` to match the image size, framerate, and format of what `usb_cam` publishes.
- In the second terminal I'm running `rqt_image_view`.
- In the third terminal I'm running a `rosimagesrc` pipeline:
  - `rosimagesrc ros-topic=/usb_cam/image_raw ! queue ! videoconvert ! xvimagesink`

| Video Source | rqt_image_view | rosimagesrc |
| -- | -- | -- |
| usb_cam | 30 FPS | 2 FPS |
| rosimagesink | 30 FPS | 30 FPS |

I used `ros2 topic info -v /usb_cam/image_raw` to inspect the publishers/subscribers, and found:
- The subscriber QoS settings are identical between `rosimagesrc` and `rqt_image_view`.
- The publisher QoS settings between `rosimagesink` and `usb_cam` only differ in one way: `rosimagesink` uses a queue depth of 5 while `usb_cam` uses a queue depth of 100 (this looks [hardcoded](https://github.com/ros-drivers/usb_cam/blob/1091c5b3437d8ea4d04fa8c1593f648001daf539/src/usb_cam_node.cpp#L49)).

I also tried switching between the publishers without restarting the subscribers to see if there was any difference. The only difference I saw was that the endianness is different between the publishers. I honestly don't know if the endianness field in the message in is meaningful, since I don't see the `is_bigendian` value set explicitly anywhere in the `usb_cam` node sources.

A few other things I've tried:
- I tried increasing the socket receive buffer size to 50 MB up from 10 and did not notice any change.
- I tried only running `rosimagesrc` (not `rqt_image_view`) to see if it was an issue with the two programs competing for resources somehow. I did not notice any change.

The issue also seems really intermittent: when I tried to record it, it was initially not working but after switching a few times it started working fine?

https://github.com/user-attachments/assets/9f71a38e-e3ed-4bc9-9976-26463ead3df6

Do you have any suggestions as to how I could narrow down why `rosimagesrc` works with the `rosimagesink` publisher but not with the `usb_cam` publisher?

breqdev, 2025-03-14:
Another discovery: Switching middleware entirely from `rmw_cyclonedds_cpp` to `rmw_zenoh_cpp` doesn't change anything about the behavior (I still get the same 2 FPS video feed with `rosimagesrc` and 30 FPS feed with `rqt_image_view`). This would seem to indicate it's not a DDS issue at all?

StefanFabian, 2025-05-11:
@BrettRD if you want you can take the fix from our fork: https://github.com/tu-darmstadt-ros-pkg/ros-gst-bridge
I can also make a PR with all of the improvements I have made. I don't have the time to separate it into single functionality PRs, but if a single PR is fine with you, I would open one.